raspberrypi/hats #20: GPIO header

disk91, 2015-02-23:
Could you add some link to find the 10-12mm GPIO header, they are really hard to find

jasaw, 2015-04-08:
Part Number: 929975-01-20
Manufacturer: 3M Electronic Solutions Division
http://au.mouser.com/Search/ProductDetail.aspx?R=929975-01-20virtualkey51750000virtualkey517-929975-01-20

Part Number: PPPC202LFBN-RC
Manufacturer: Sullins Connector Solutions
http://www.digikey.com/product-search/en?x=0&y=0&lang=en&site=us&keywords=S7123-ND

disk91, 2015-04-19:
Not sure about the second one : the high seems to be 8.5mm not following the expected standard. Isn't -it ?

jasaw, 2015-04-20:
From the HAT specifications: "uses spacers between 10mm and 12mm."

The HAT connector height of 8.5mm + roughly 2mm plastic base on the male header the Pi side = ~11mm.
If you get an 11mm spacer, that should fit nicely.

11mm spacer:
http://au.mouser.com/Search/ProductDetail.aspx?R=M1258-2545-SSvirtualkey68520000virtualkey761-M1258-2545-SS

disk91, 2015-04-20:
Ok makes sense.
It means that the documentation should be clarified to indicate the 2mm of plastic have to be taken in consideration my understanding was 11mm of plastic on the header side.

lurch, 2015-12-08:
https://www.raspberrypi.org/blog/introducing-raspberry-pi-hats/ also contains some links to suitable connectors.

mcintoda, 2016-01-02:
Where is the electrical pin out for the 40 pin connector specified?

lurch, 2016-01-03:
> Where is the electrical pin out for the 40 pin connector specified?

I dunno what level of detail you're looking for, but here's some useful pointers:
https://www.raspberrypi.org/documentation/usage/gpio-plus-and-raspi2/README.md
https://www.raspberrypi.org/documentation/hardware/raspberrypi/gpio/README.md
https://www.raspberrypi.org/wp-content/uploads/2014/04/bplus-gpio.png
https://www.raspberrypi.org/documentation/hardware/raspberrypi/schematics/Raspberry-Pi-B-Plus-V1.2-Schematics.pdf
https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2835/BCM2835-ARM-Peripherals.pdf
https://www.scribd.com/user/105980770/gvanloo8080

http://elinux.org/RPi_Low-level_peripherals
http://elinux.org/RPi_BCM2835_GPIOs
http://pinout.xyz/
http://www.mosaic-industries.com/embedded-systems/microcontroller-projects/raspberry-pi/gpio-pin-electrical-specifications