raspberrypi/hats #23: HAT definition applies to A+ as well as B+ (Right?)

unsureofmyself, 2015-03-28:
hats/Readme.md makes no mention of the Raspberry Pi model A+, only the B+.
I understand that the HAT definition applies to the A+ model too, and the Readme.md ought to reflect this.
Perhaps the same is true of the newer Pi 2 as well?
Thanks,
JC.

lurch, 2015-12-08:
The way I understand it, the "HAT definition" is only concerned with the add-on boards themselves; and all the Pis released so far with the 40-pin header (B+, A+, B2, Zero) are HAT-compatible.
The README.md only specifically mentions the Raspberry Pi B+, because that was the first board released with the 40-pin header.

IMHO it makes no sense to update the hats README each time a new Raspberry Pi model is released (but maybe the B+ could be de-emphasised, now there are multiple HAT-compatible Pis).

RandomlyKnighted, 2016-02-15:
@unsureofmyself @lurch Take a look at the PR I've made. Do the changes I've made to the README clarify the HAT definition enough?

lurch, 2020-03-17:
Fixed by https://github.com/raspberrypi/hats/commit/1f2fbdf09df8c0c63584c633cd8b6e04bf53f841 ?

JamesH65, 2020-03-17:
I believe so, OK to close?

jimbojr, 2020-03-20:
Yes

jimbojr, 2020-03-20:
Yes

closed 2020-03-20 by pelwell