raspberrypi/hats #28: How may be identified more than one HAT board when stacked?

jgrulich, 2015-05-28:
How may I2C read more than one HAT EEPROM if there is only one fixed 0x50 I2C address?
Even there is unique UUID, the hardware EEPROM address allows to read only one EEPROM.
If stacked two boards, the address collision results in global communication error and no one is recognized.

P33M, 2015-05-28:
This is intentional behaviour, designed into the specification such that any collision between data stored in EEPROMs with identical addresses (such that might arise when a user "stacks" two or more HATs) is identifiable in all cases.

Automatic HAT setup and probing is limited to a single board only. As GPIO setup is an integral function performed when a HAT is probed for use, the decision was made when designing the specification to not support the case where more than one board was attached to the GPIO header, as resolving GPIO conflicts would be extremely complex and would likely lead to obscure and user-unfriendly breakage of random segments of functionality if two arbitrary HATs were simultaneously used.

There is no specific exclusion in the specification for designing stackable boards - if boards with stacking capability are available from a manufacturer, then there are options available for manual configuration, such as providing a custom DT overlay and disabling the onboard EEPROMs.

closed 2015-05-28 by P33M

jgrulich, 2015-05-29:
Understand. The idea is good, but for such a identification is better to use some other memory with unique SW addressing. Than the address may be, for example,  UUID which will every be unique.
In my case this is quite important, because I'm using several stacked boards with different functions.
Another, more important, issue is that I have several functions on one board and need to put several overlays into one EEPROM file and even more need to add the additional setup for these overlays.
Exactly I need to put there for example this:

dtoverlay=w1-gpio
dtoverlay=i2c-rtc,pcf8563
dtoverlay=iqaudio-dacplus
dtoverlay=lirc-rpi,gpio_in_pin=25,gpio_in_pull=up

Another question is why is not possible to use only the text explanation of the overlays included the additional setup, as mentioned above, instead of full overlay files which are quite big and need long time to be up/down loaded. BTW the idea to have the setup only in the EEPROM is not possible, because instead of overlay is needed at least driver.

Point for improvement:
add the option to use only the text expression like in the config.txt instead of overlay file.