raspberrypi/hats #33: RPI does not boot if HAT EEPROM is corrupted

hifiberry, 2016-03-08:
I've seen a problem with a board where the EEPROM content wasn't written correctly. With older 3.x kernels the board boots, but not with the latest kernel/firmware. Is there any kernel parameter to ignore the ID PROM contents? This needs to be only temporary to write the correct EEPROM contents.

pelwell, 2016-03-09:
You can disable the EEPROM reading using the config.txt setting force_eeprom_read=0. I'll take a look at the reading code to see why it doesn't boot. If you can read the EEPROM contents and upload it somewhere (or email it to me) I can use it as a test case.

hifiberry, 2016-03-09:
Thanks, will test this.
I will see that I can find a card with this problem and send you en EEPROM dump.

mlyle, 2016-06-18:
Not corrupted, but here's a .EEP that fails to boot on Pi2.  If I comment out the alternate functions GPIO lines it works.

(Had to zip this to attach it, even though it actually expands the file :P )
[doesntwork.zip](https://github.com/raspberrypi/hats/files/321496/doesntwork.zip)