raspberrypi/hats #34: HAT not compatible with official RPi enclosure

AnttiLukats, 2016-04-28:
is there is new HAT spefication? Because any HAT design to the old one would not fit into the RPi official enclosure :(

RandomlyKnighted, 2016-04-28:
I haven't heard of any new HAT specification. As far as I know the official Raspberry Pi enclosure is meant to hold only the Pi.

medo64, 2016-05-13:
Is it allowable to make additional cutouts on the HAT that would allow it to fit into the official case; i.e. would board with additional cutouts on side still qualify as HAT?

Strictly reading specification, it would seem no as it does say "board must be 65x56 mm as per this drawing"

RandomlyKnighted, 2016-05-13:
@medo64 Well technically the board is still 65x56 if it has a cutout in it, because the sides full length would meet the specs.

medo64, 2016-05-13:
@RandomlyKnighted True, although part "per this drawing" part muddies the waters a bit. :)