raspberrypi/hats #43: EEPROM emulation.

diymat, 2017-04-16:
Hi. I have a question and think that many people will have the same problem as well. I understand that if a HAT does not not have its own micro - EEPROM is needed. But at the moment I develop couple of HATs with their own ARM uCs. I can emulate the full EEPROM functionality using uC I2C interface & flash memory. Can I do it, or I still need a separate EEPROM IC? Timing is not an issue as the uC is ready much earlier than RPI starts reading the content form it.

jimbojr, 2017-05-02:
Happy to allow a micro to provide the data - yes this is fine as long as the Pi can't tell the difference!

JamesH65, 2017-05-02:
@diymat - OK to close?

diymat, 2017-05-18:
Yes, it is OK to close, thank you, Sorry. After my cataract operation unfortunately I have infected my eye and was unable to use computers :). It

closed 2017-05-18 by diymat