raspberrypi/hats #45: Licence missing from the project

attila16881, 2017-07-18:
Hi All,
I was looking for the specifications but I cannot find any reference to licence or IP of the HATS specifications/requirements, EEPROM design requirements, etc.

Could you please clarify and also add it to the project page? 

Thanks!

nicolalunghiemutex, 2017-09-01:
any news on this?

JamesH65, 2017-09-01:
@jimbojr Can you comment on this?

jimbojr, 2017-09-01:
I'm afraid I don't understand the question - what exactly is missing?
EEPROM requirements are specified in the design guide https://github.com/raspberrypi/hats/blob/master/designguide.md

nicolalunghiemutex, 2017-09-01:
Hi
the license for the source code of the tools.
Without it as stated here 
https://www.infoworld.com/article/2615869/open-source-software/github-needs-to-take-open-source-seriously.html

"Code without an explicit license is protected by copyright and is by default All Rights Reserved. The person or people who wrote the code are protected as such. Any time you're using software you didn't write, licensing should be considered and abided."

So practically if I want to modify/improve the code, add extension, fork it and so on I cannot do that.

Regards
Nicola Lunghi

JamesH65, 2017-09-04:
Which tools are you specifically talking about? The Linux kernel is clearly open source GPL licenced so that is in the clear.

nicolalunghiemutex, 2017-09-04:
the eepromutils software

https://github.com/raspberrypi/hats/tree/master/eepromutils

https://github.com/raspberrypi/hats/blob/master/eepromutils/eepdump.c
https://github.com/raspberrypi/hats/blob/master/eepromutils/eepflash.sh
https://github.com/raspberrypi/hats/blob/master/eepromutils/eeptypes.h
https://github.com/raspberrypi/hats/blob/master/eepromutils/eepmake.c

etc etc

JamesH65, 2017-09-04:
Ah, OK, I'll try and get that sorted. Although you are perfectly entitled to use them, just that redistribution rights are not particularly clear.

JamesH65, 2017-09-04:
There should now be a license file in there, can you check it is adequate for your needs?