raspberrypi/hats #46: Change Request: Add optional previsions to the Spec for stacking HATs

tsayles, 2017-10-04:
There are use cases and there is value in the ability to stack multiple HATs.  
As standards should not be written in stone, but ought to evolve over time, it should be appropriate for members of the community to develop (through backwards compatible extensions) methods for the discovery and configuration of multiple stacked HATs. aka "HAT++"???

To that end I would invite the HATs contributors and the community at large to discuss [within this issue] how such a stacking extension could be developed and a path for that extension to be included in a revision to the original HAT specification.

shawaj, 2017-10-20:
That is a great idea, especially in light of the new "phat stack" product from Pimoroni

transistorgrab, 2017-10-22:
The identification part of this would be easy enough, just configure the I²C EEPROM for another address. But this needs to be user configurable. Maybe one kind of module can be stacked multiple times.
The difficulties come, where two HATs drive to the identical GPIO. Proven practise is that at each output a series resistor has to be provided, even if only for signal termination. 
But how can one make sure that each output to one GPIO has the same drive strength and will not be damaged by a stronger driver? Possibly by limiting the max. drive current to 1 mA?

Stacking is a great idea. But I think there are a lot of problems to solve when combining existing HATs with such designed for stacking.
I think only modules designed to a specification with stacking in mind can be allowed to be used for stacking.

From the hardware/mechanical point of view a 0.1'' socket strip with long pins (i.e. https://www.adafruit.com/product/2223) is the simplest solution for this purpose.

shawaj, 2017-10-22:
The capability to just load multiple HAT EEPROM data files and device trees would probably be enough and then just say "everything else is at your own risk and look at pinout.xyz for compatibility"?

jimbojr, 2017-10-25:
Whether we should (could) stack HATs was extensively discussed both internally and publicly on the forum when we were in the process of creating the HAT spec.

In short it is too complex to guarantee that stacking any HAT on another won't cause either Pi or HAT damage (or both). Hence there is no support for stacking.

That doesn't mean you cant stack HATs - it just means we don't support or recommend it unless you really know what you are doing.

closed 2017-10-25 by jimbojr