raspberrypi/hats #50: No run/reset connector on mechanical drawings for RPi 3 B+

JDat, 2018-03-30:


JamesH65, 2018-03-30:
@Roger-Thornton

lurch, 2020-03-17:
https://github.com/raspberrypi/hats#faq says that a board connecting to the run/reset pins doesn't conform to the HAT spec (which is what this repo is all about).

Does https://github.com/raspberrypi/documentation/tree/master/hardware/raspberrypi/mechanical contain any of the info you're looking for? (note that it's in a different repo!)

JDat, 2020-03-17:
Let's say i am designing HAT compatible board for me with watchdog function. So I need run/reset location. Is doesn't matter will it be called 100% HAT compatible. Also... It is possible to use pogo pins on HAT and get run/stop connection without soldering on RPi board.

BTW post is already 2 years old. Many things changed from that time...

lurch, 2020-03-17:
> Is doesn't matter will it be called 100% HAT compatible.

From my reading of the HAT specs, a board is either "a HAT" (meets all the guidelines) or "not a HAT" (doesn't meet all the guidelines), there's no such thing as "not 100% HAT compatible".

You're of course free to add pogo-pins to any add-on board you design, but you can't call it a HAT, and AFAIK there's no guarantee that the location of the run / reset connectors won't be moved on different / future models of Raspberry Pi.

Disclaimer: my views may not reflect those of Raspberry Pi Trading, and I'm only a software-guy not a hardware-guy :wink:

JamesH65, 2020-03-17:
@jimbojr Any thoughts? I have no idea....