raspberrypi/hats #58: eepflash.sh still advising rpi-update in two locations

Aydan, 2019-04-17:
eepflash.sh still advises to do rpi-update in case `dtoverlay i2c-gpio i2c_gpio_sda=0 i2c_gpio_scl=1` or `modprobe at24` fails. I think this is outdated information.

pelwell, 2019-04-17:
I assume by "outdated" you mean that upgrading the Raspbian kernel package would also get a compatible kernel, and that is the preferred method?

Personally, I still prefer rpi-update - it's easier to pick firmwares and downgrade, and is guaranteed not to cause any other packages to be modified, either correctly or as a result of finger trouble.

Aydan, 2019-04-17:
This might be so, but is ist relly necessary nowadays do get the functionality required for eepflash.sh?
If not, then advising a hat developer to rpi-upgrade if it's not necessary might break his system with a beta firmware/kernel.
Also the consensus on the forums seems to be "Don't use rpi-flash unles you know exactly what you're doing or someone in the know tells you to to fix a specific bug."

pelwell, 2019-04-17:
Remember that this is a message only displayed in the event of an error. Have you seen the message in the case that the kernel was already new enough to not need upgrading?

Not having a sufficiently new kernel to perform a required operation is a good reason to upgrade. I would never advise anyone to upgrade, by whatever means, if there system was meeting their requirements.

Aydan, 2019-04-17:
I haven't seen the message when running the tool, but I did read the code.
So the first recourse to an error due to missing modules or overlays should be to upgrade with apt. But the message for the missing overlay only refers to rpi-update and the other one lists rpi-update first..
And since the current kernel and firmware packages have all required files, why refer a possibly unknowing user to a possibly unsafe upgrade path when the official upgrade path will solve the problem?

pelwell, 2019-04-17:
Give us a Pull Request and we'll consider it.

lurch, 2019-08-05:
pinging @andrum99 given all his similar PRs on the documentation repo recently :slightly_smiling_face: