raspberrypi/hats #59: Dimension error in uhat-board-mechanical.pdf

glansberry, 2019-05-02:
The board overall dimension is called out as 30.5mm,  this does not match the board shape of the rpi zero of 30mm.  Also if you add up the the dimension of mounting holes to edge of board (3.5mm + 3.5mm) plus dimensioned space between hole centers of 23mm it does not add up to 30.5mm.  This makes me pretty confident that the 30.5mm dimension is incorrect.

If changing the drawing, it would also be nice to indicate where pin 1 of the 40 pin header is.

JamesH65, 2019-05-03:
@jimbojr @Roger-Thornton

lurch, 2019-08-05:
There's some 90-degree text on the uHAT PDF which says
```
30.5mm FOR SMT STYLE CONNECTOR
OTEHRWISE 30mm FOR THROUGH HOLE CONNECTOR
```
(typo is in the PDF, not mine :wink: )

JamesH65, 2019-08-06:
@jimbojr @Roger-Thornton Ping.