raspberrypi/hats #60: PoE

nbxmike, 2019-06-10:
Fixes:
* Delete PoE comment from uHat
* Add PoE comment to Hat
Suggestions:
* Move the keep out area on the uHat specification to actually cover the RUN/TV connectors 
* Add the RUN connector location to the Hat specification
Curious:
* Why does the PoE location on the Hat specification not match the PoE location on the RPi (or PoE Hat for that matter)?

lurch, 2019-08-05:
The RUN connector is in different places on different models of Raspberry Pi, whereas the HAT and uHAT specs are universal across all models of Raspberry Pi. (AFAIK it's perfectly valid to use a HAT with a Pi Zero, or use a uHAT with a Pi 4B)

nbxmike, 2019-08-06:
* If it is worthwhile to recommend "BOARD MUST MAKE SURE PoE HEADER DOES NOT FOUL" on the uHat spec then it is worthwhile to say the same on the Hat specification.
* It would be nice to stop moving the RUN connector and specify it, someday.
* It would seem logical that the POE location in the Hat specification at least match one real design.

JamesH65, 2019-08-06:
@jimbojr @Roger-Thornton Any comment?

lurch, 2019-08-06:
> If it is worthwhile to recommend "BOARD MUST MAKE SURE PoE HEADER DOES NOT FOUL" on the uHat spec then it is worthwhile to say the same on the Hat specification.

It _does_ mention that at https://github.com/raspberrypi/hats/blob/master/README.md#hat-requirements (although not directly on the HAT PDF).