raspberrypi/hats #61: eepmake: How to access the flashed custom file from user space?

taudac, 2019-08-12:
After flashing a custom file to the ID EEPROM, how do you actually access it later from user space? I though it would be in `/proc/device-tree/hat/...`, but it is not.

pelwell, 2019-08-12:
It ought to turn up under `/proc/device-tree/hat/custom_<n>`, where `<n>` is a decimal integer counting up from 0.

Can you post your eeprom.txt file?

taudac, 2019-08-12:
The issue is no longer reproducible. I am sorry for the noise.

I am building the EEPROM image with:

    eepmake myhat-eeprom.txt myhat.eep myhat.dtbo -c myhat-params.txt

The `myhat-eeprom.txt` file contains no custom data related entries.

I am now able to access the data via `/proc/device-tree/hat/custom_0`.

pelwell, 2019-08-12:
Cool - you may be the first person to use it in anger.

taudac, 2019-08-12:
@pelwell Many thanks for your very quick response. It has shown me that I must be doing something wrong.

closed 2019-08-12 by taudac

pelwell, 2019-08-12:
If you fail to get it working, upload myhat.eep somewhere and I'll have a look.