raspberrypi/hats #62: Dimension error in hat-board-mechanical.pdf

matthiesenj, 2019-08-20:
If the HAT is 56.5mm wide (as shown on the drawing), the distance from hole center to edge should be 4mm, not 3.5mm. If it's 56mm wide (as mentioned as an option in the text), the hole distance is correct, though.
Pi 3 and 4 have 56mm wide boards (don't know about older models) , so maybe the 56.5mm option should just be dropped altogether?

jonhp, 2019-11-11:
The vertical note along the right vertical edge says that the 56.5mm width is to accomodate smd headers. ie there is an additional 0.5mm of vertical size allowed. the 3.5mm from the bottom edge to the mounting hole centerline added to the 49mm vertical distance between the mounting holes accurately positions them from the bottom edge. so can you further explain your concern or should this issue be closed?

JamesH65, 2019-11-11:
@jimbojr Is this one for you?

matthiesenj, 2019-11-11:
> The vertical note along the right vertical edge says that the 56.5mm width is to accomodate smd headers. ie there is an additional 0.5mm of vertical size allowed. the 3.5mm from the bottom edge to [...]

I read it as 3.5mm distance from hole to board edge for all holes/edges since it's only specified for one of them, and then arbitrarily focused on the bottom edge/holes which then were in violation for 56.5mm width. I suppose the drawing isn't wrong, but also not 100% explicit.