raspberrypi/hats #63: Question: Changes to power requirements for RPi 4?

jacobq, 2019-08-20:
When the RPi4 was released I remember reading that one of the reasons for changing the power connector to USB-C was to get a little more available power. If I want to make a HAT that supplies (+5VDC) power to an RPi 4, how much current does it need to be able to source? Is it still about 2A or does the new model need more?

YANG-Mingcong, 2020-07-21:
3A... I'm making my own PoE Hat for Rpi 4B.
And I'm confused cause the official PoE HAT only provides 2.5A, but it says it could compatible for Rpi4...

jacobq, 2020-07-21:
> 3A... I'm making my own PoE Hat for Rpi 4B.
> And I'm confused cause the official PoE HAT only provides 2.5A, but it says it could compatible for Rpi4...

I believe that under most circumstances the RPi (even RPi4) will not draw that much current. For example, each USB port is specified to be able to supply a certain amount of power for peripherals, so if you're not using them (or using them to the fullest) then the actual amount of current required from the supply will be less. See also https://www.raspberrypi.org/documentation/hardware/raspberrypi/power/README.md