raspberrypi/hats #76: eepflash.sh fails with ~ in eeprom filename

beikeland, 2020-12-06:
not quite sure if its a real problem, but  using ~ is a convenient short hand at least.

doesn't work

```
pi@raspberrypi:~ $ sudo ~/hats/eepromutils/eepflash.sh -w -t=24c32 -d=0 -a=50 \
>     -f=~/hats/eepromutils/eeprom.cs0.irq0.bin
This will attempt to talk to an eeprom at i2c address 0x50 on bus 0. Make sure there is an eeprom at this address.
This script comes with ABSOLUTELY no warranty. Continue only if you know what you are doing.
Do you wish to continue? (yes/no): yes
Writing...
dd: failed to open '~/hats/eepromutils/eeprom.cs0.irq0.bin': No such file or directory
Closing EEPROM Device.
Error doing I/O operation.
```

does work
```
pi@raspberrypi:~ $ sudo ~/hats/eepromutils/eepflash.sh -w -t=24c32 -d=0 -a=50 \
>     -f=/home/pi/hats/eepromutils/eeprom.cs0.irq0.bin
This will attempt to talk to an eeprom at i2c address 0x50 on bus 0. Make sure there is an eeprom at this address.
This script comes with ABSOLUTELY no warranty. Continue only if you know what you are doing.
Do you wish to continue? (yes/no): yes
Writing...
1024 bytes (1.0 kB, 1.0 KiB) copied, 4 s, 0.3 kB/s
2+1 records in
2+1 records out
1403 bytes (1.4 kB, 1.4 KiB) copied, 5.39316 s, 0.3 kB/s
Closing EEPROM Device.
Done.
```

mhei, 2021-06-02:
I think this is expected behavior and not a bug: since you pass the parameters to sudo, the script then runs as user root and it tries to locate the eeprom.cs0.irq0.bin in the root user's home directory and referenced path. I don't see how this could be prevented.

pelwell, 2021-06-22:
Having rediscovered this issue I've added support for passing the file name as a separate parameter, e.g.
`./eepflash.sh -f ~/my_hat.eep ...`. This allows shell expansion (and TAB completion) to work on it.